SmingHub/Sming #2784: #warning "**Y2038** time_t is only 32-bits in this build configuration stops the build for esp32 due to -Werror=cpp

xeonqq, 2024-06-01:
When using the lastest sming in docker and try to build esp32
I have the following error:
[image]

Have to delete that line from DataTime.cpp to make it work, I don't know where this -Werror=cpp is defined. Or if this warning is necessary.

mikee47, 2024-06-01:
See issue #2758.

1. Build with `STRICT=1` to allow compilation to continue
2. Fix the problem. i.e. you're using an old IDF version (4.3 or 4.4)

xeonqq, 2024-06-01:
I just rebuild the docker image with the [Dockerfile](https://github.com/SmingHub/Sming/blob/develop/Tools/Docker/cli/Dockerfile), cuz I want the new websocket fix. Shouldn't it pull the latest IDF version automatically for me?

mikee47, 2024-06-01:
Ah, I see. The default IDF installed is version 4.4, we should probably update that to 5.2.

mikee47, 2024-06-01:
For now, you can modify the dockerfile as per https://sming.readthedocs.io/en/latest/_inc/Sming/Arch/Esp32/README.html#idf-versions:

```
RUN INSTALL_IDF_VER=5.2 Tools/install.sh $INSTALL_ARGS
```

mikee47, 2024-07-03:
Warning now removed.

closed 2024-07-03 by mikee47