SmingHub/Sming #2785: make flash does not upload modified javascript file

xeonqq, 2024-06-01:
If I modify a javascript file which is included in index.html, after calling ```make flash```, the old file is not updated in the MCU.
The only way to make it really update that file is to change the file name of this javascript file, and do a ```make clean``` then ```make flash```. Quite time consuming during development.

mikee47, 2024-06-01:
Your issue is somewhat lacking in detail, but I'll make a few observations. Note that the build system isn't smart enough to detect changes in the target filesystem layout so you must do this manually.

In general when you need to rebuild a filing system partition image use `make buildpart`. If the partition in question is called, say, `spiffs0`, then you can be more specific and use `make buildpart PART=spiffs0`. That will rebuild the image from the source files.

Regarding `make flash`, if you run `make help` you'll see it described as `Write the boot loader and all defined partition images`. By contrast, `make flashpart PART=spiffs0` will flash only the `spiffs0` partition.

If you're doing a lot of this sort of iterative development on high-level stuff like web interfaces, you might consider getting your project building for Host and use that for development (if you're not already!). It's much faster than repeatedly re-flashing devices!

xeonqq, 2024-06-02:
Thanks for the hint. I tried the make buildpart, it is still not clever enough to detect the change and build the newly modified file, I still need to change the file name so that it can detect the change.
I do need to try the host build.

closed 2024-06-02 by xeonqq