SmingHub/Sming #2816: can't build for Arch host on Fedora 39

pljakobs, 2024-06-18:
This is probably none of Sming's fault, but since I've just experienced it, I thought I'd document it here:
After updating from Fedora 38 to 39, it seems I cannot build for the Host architecure anymore. 
```
$ make SMING_ARCH=Host
/opt/Sming/Sming/build.mk:227: *** Compiler '' not recognised. Please install GCC tools.. 
```
I found a few pointers that might be connected to this: 
https://discussion.fedoraproject.org/t/fedora-39-issues-compiling-for-32-bit-on-a-64-bit-system-with-gcc/95954

So, I'm not really sure if the error I'm seeing (basically failure to select a compiler) is something in the Sming build system or just an issue in F39. 

glibc and libgcc installed as .i686 packages.

mikee47, 2024-06-18:
What do you get with `gcc -v` ?

pljakobs, 2024-06-18:
```
$ LANG=C gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,m2,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-13.3.1-20240522/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-offload-defaulted --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.1 20240522 (Red Hat 13.3.1-1) (GCC) 
```

mikee47, 2024-06-19:
The logic for compiler version detection changed in #2798 to support clang. Previously we parsed `gcc -dumpversion` but this is now `gcc -v` so we can get the compiler name as well.

This happens in `Sming/build.mk` line 214:

```
$(CC) -v 2>&1 | $(AWK) -F " version " '/ version /{ a=$1; gsub(/ +/, "-", a); print a, $2}'
```
It should produce `gcc 13.3.1 20240522 (Red Hat 13.3.1-1) (GCC)` for you.

If either `CC` or `AWK` are incorrect then this filter produces an empty string (rather than perhaps a more useful error message). The logic here could do with a little improvement I think.

Anyway, to debug this, try adding this line just before it:

```
$(info CC = $(CC))
$(info AWK = $(AWK))
$(info $(shell $(CC) -v))
```

pljakobs, 2024-06-19:
I think I found it! My setup is using German language so the version Line is "gcc-Version 13.3.1 20240522 (Red Hat 13.3.1-1) (GCC)" whereas in the English language message, "version" is lower case.

so, while ```make SMING_ARCH=Host``` fails to identify the compiler, ```LANG=C make SMING_ARCH=Host``` works. 

here's the output of gcc -v 
```
$ gcc -v
Es werden eingebaute Spezifikationen verwendet.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Ziel: x86_64-redhat-linux
Konfiguriert mit: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,m2,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-13.3.1-20240522/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-offload-defaulted --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread-Modell: posix
Unterstützte LTO-Kompressionsalgorithmen: zlib zstd
gcc-Version 13.3.1 20240522 (Red Hat 13.3.1-1) (GCC) 
```
and here of LANG=C gcc -v
```
$ LANG=C gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,m2,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-13.3.1-20240522/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-offload-defaulted --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.1 20240522 (Red Hat 13.3.1-1) (GCC) 
```

I suggest using LANG=C for compiler detection, as there are certainly other situations with non English language where this will fail.

```
# Detect compiler version and name
DEBUG_VARS                              += COMPILER_VERSION_FULL COMPILER_VERSION COMPILER_NAME
COMPILER_VERSION_FULL   := $(shell LANG=C $(CC) -v 2>&1 | $(AWK) -F " version " '/ [Vv]ersion /{ a=$$1; gsub(/ +/, "-", a); print a, $$2}')
COMPILER_NAME                   := $(word 1,$(COMPILER_VERSION_FULL))
COMPILER_VERSION                := $(word 2,$(COMPILER_VERSION_FULL))
```

mikee47, 2024-06-19:
Thank you! Good catch, will get that fixed.

closed 2024-06-22 by mikee47