SmingHub/Sming #2819: ESP32 builds are massive

mikee47, 2024-06-19:
HostTests builds for esp32 look suspicously large. These are all default builds. Needs further investigation.

esp32: app.bin 1,173,952 bytes

```
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC0B18|   12216
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F447910|  293104
       bss|      Uninitialized Data (RAM)|    3FFC0B18|    3FFC4F48|   17456
 iram_text|            Cached Code (IRAM)|    40080400|    40094148|   81224
      text|           Uncached Code (SPI)|    400D0020|    4018FF84|  786276
```

esp32c3: app.bin 1,207,664 bytes

```
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FC8FE00|    3FC926E8|   10472
    rodata|           ReadOnly Data (SPI)|    3C0D0020|    3C116AE0|  289472
       bss|      Uninitialized Data (RAM)|    3FC926F0|    3FC96C50|   17760
 iram_text|            Cached Code (IRAM)|    40380250|    4038FE00|   64432
      text|           Uncached Code (SPI)|    42000020|    420CD1BC|  840092
```

esp8266: rom0.bin 771,584 bytes

```
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8794|    1940
    rodata|           ReadOnly Data (RAM)|    3FFE87A0|    3FFE9AF0|    4944
       bss|      Uninitialized Data (RAM)|    3FFE9AF0|    3FFEFFC8|   25816
      text|            Cached Code (IRAM)|    40100000|    40106B8F|   27535
irom0_text|           Uncached Code (SPI)|    40202010|    402B5F58|  737096
```

rp2040:  app.bin 710,172 bytes

```
     Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
        data|        Initialized Data (RAM)|    200000C0|    200015E4|    5412
         bss|      Uninitialized Data (RAM)|    20001608|    2000CB48|   46400
flash_binary|           Uncached Code (SPI)|    10000000|    100AD61C|  710172
```

mikee47, 2024-06-23:
Comparing build logs between IDF 4.4 and IDF 5.2:

```
tests/HostTests
  log                data   rodata  bss    iram_text  text     Total Used RAM  Free RAM  Free IRam
  -----------------  -----  ------  -----  ---------  -------  --------------  --------  ---------
  log-esp32-4.4.txt  10592  248088  16336  78800      683690   26928           308944    52272    
  log-esp32-5.2.txt  12216  293572  17472  81224      788196   29688           306184    49848    
  Difference         +1624  +45484  +1136  +2424      +104506  +2760           -2760     -2424    
```

mikee47, 2024-06-24:
IDF 5 adds `esp_newlib_init_global_stdio` which results in various I/O code from newlib being linked. Not a lot though.

mikee47, 2024-06-24:
Building with WPA3 disabled reduces size:

IDF 4.4:

ESP_WIFI_ENABLE_WPA3_SAE=n

```
HostTests: Linking out/Esp32/esp32/debug/build/app.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC0494|   10548
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F43BCD4|  244916
       bss|      Uninitialized Data (RAM)|    3FFC0498|    3FFC43F8|   16224
 iram_text|            Cached Code (IRAM)|    40080400|    400937D0|   78800
      text|           Uncached Code (SPI)|    400D0020|    4016EC86|  650342
Total Used RAM : 26772
Free RAM : 309100
Free IRam : 52272
```

IDF 5.2:

ESP_WIFI_ENABLE_WPA3_OWE_STA=n
ESP_WIFI_ENABLE_WPA3_SAE=n
ESP_WIFI_SOFTAP_SAW_SUPPORT=n

app.bin: 1,077,520 bytes

```
HostTests: Linking out/Esp32/esp32/debug/build/app.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC0AF8|   12184
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F43ED5C|  257340
       bss|      Uninitialized Data (RAM)|    3FFC0AF8|    3FFC4E28|   17200
 iram_text|            Cached Code (IRAM)|    40080400|    400940F0|   81136
      text|           Uncached Code (SPI)|    400D0020|    40181308|  725736
Total Used RAM : 29384
Free RAM : 306488
Free IRam : 49936
```

mikee47, 2024-06-24:
Release builds are very similar between IDF 4.4 and 5.2. Conclusion: More debug code/text in IDF 5.2.

IDF 4.4

```
HostTests: Linking out/Esp32/esp32/release/build/app.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC04B8|   10584
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F435EF8|  220888
       bss|      Uninitialized Data (RAM)|    3FFC04B8|    3FFC4338|   16000
 iram_text|            Cached Code (IRAM)|    40080400|    40093008|   76808
      text|           Uncached Code (SPI)|    400D0020|    40171DDE|  662974
```

IDF 5.2

```
HostTests: Linking out/Esp32/esp32/release/build/app.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFBDB60|    3FFC0AB8|   12120
    rodata|           ReadOnly Data (SPI)|    3F400020|    3F438034|  229396
       bss|      Uninitialized Data (RAM)|    3FFC0AB8|    3FFC4C98|   16864
 iram_text|            Cached Code (IRAM)|    40080400|    40093A9C|   79516
      text|           Uncached Code (SPI)|    400D0020|    40172B24|  666372
```

pljakobs, 2024-10-29:
is there a way to reduce the amount of extra code in non-release builds? in my case, it looks like the application won't fit into a 1MB segment when built with SMING_RELEASE=0 which limits my ability  to debug on the Esp32 platform (I assume with a release build, gdb with a hw debugger won't be of much use either as symbos are stripped?)

pljakobs, 2024-10-29:
just noticed that even with a release build, the rom image is >1.2MB whereas for the 8266 it's <900kB. I guess that's the price to pay. 
It's still true that the ROM image needs to fit into one 1MB page, right?

mikee47, 2024-10-29:
You could try mucking around with some of the IDF settings, see if that makes a difference. Also try reducing debug verbosity as debug messages take up space. For example:

```
# Rebuild everything with no debug output
make clean components-clean
make -j DEBUG_VERBOSE_LEVEL=0
# Rebuild application plus selected components with increased debug output
make -j DEBUG_VERBOSE_LEVEL=2 
```

Note that symbols aren't stripped in any build versions since this information never gets into the firmware images. Building with `ENABLE_GDB=1` does affect the optimisation level though (uses `-Og`) since debugging optimised code is generally problematic.

mikee47, 2024-10-29:
The esp32 series have better virtual memory management hardware and aren't limited to 1MByte AFAIK.