SmingHub/Sming #2868: *** Preparation for Release 6.0.0 ***

slaff, 2024-07-19:
This issue is created to collect all the tasks we need to do before the 5.2 release.

### Source Code
- [ ] Update Sming/Core/SmingVersion.h.
- [ ] Update Readme.md.
- [ ] Merge develop into master.
- [ ] Tag the source code in the master branch.
- [ ] Add upgrade document

New features
------------------

Improvements
-------------------

Bug fixes
------------

Changes
-----------

Deprecations
-----------------
- [ ] Remove official support for ESP8266 tollchains other than ESP Quick Toolchain.
- [ ] Remove official support for IDF ESP versions lower than 5.2

Breaking Changes
------------------------

Security
-----------

All PRs scheduled for this release can be seen from [here](https://github.com/SmingHub/Sming/milestone/35)

slaff, 2024-12-11:
@mikee47 do you think you can add your additions for Pico 2350 partitions and esp32s2 USB before Christmas?

mikee47, 2024-12-12:
@slaff I'm going to leave RP2350 partitions for the time being as it's going to be tricky, but the USB stuff yes I'll get that ready.

closed 2025-01-07 by slaff