SmingHub/Sming #2876: Error during compiling

teicors, 2024-08-21:
Hi,
until yesterday  I was able to compile my little project, but today I'm getting this error:

Creating SPIFFS image 'out/Esp8266/debug/firmware/spiff_rom.bin'
Traceback (most recent call last):
  File "/opt/sming/Sming/Libraries/Spiffs/spiffsgen.py", line 566, in <module>
    main()
  File "/opt/sming/Sming/Libraries/Spiffs/spiffsgen.py", line 542, in main
    raise RuntimeError('given base directory %s does not exist' % args.base_dir)
RuntimeError: given base directory files does not exist

I tried make  with clean, dist-clean, submodules-clean, components-clean but no luck at all: can someone can give me the  direction to solve my issue?

BR,
Angelo

mikee47, 2024-08-21:
Try `make V=1` to see the spiffsgen command line, may help diagnose the issue.

You didn't mention `config-clean`... But in general I just `rm -rf out` when I want to reset the application build. I'll often add `$SMING_HOME/out` but that's unlikely to help in this case.

teicors, 2024-08-21:
AARRGGHH My fault ! spiff_rom.bin !! I just added to component.mk the option for spiff, but not the dir :)
I added and fixed :)
Thanks !

closed 2024-08-31 by slaff