SmingHub/Sming #2909: dynamic allocation of Hardware_PWM fails on esp32c3

pljakobs, 2024-11-01:
my esp_rgbww firmware allocates Hardware_PWM dynamically, which works well on Esp8266 and Esp32 SoCs.

On Esp32c3 I get the following error:
```
assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)
```

tlsf block_locate_free finds, as far as I understand, a suitably large free block of memory for the requested size. 

The assert is rather sensible, the size of the block found sould be at least as large as the size requested.

My first thought was that this might be a case of type sizes again, but both pins[] as well as pin_count are defined as uint8_t, so at least from that point, it should make no difference whatsoever.

Here's block_locate_free:

```
static inline __attribute__((always_inline)) block_header_t* block_locate_free(control_t* control, size_t size)
{
	int fl = 0, sl = 0;
	block_header_t* block = 0;

	if (size)
	{
		mapping_search(control, size, &fl, &sl);

		/*
		** mapping_search can futz with the size, so for excessively large sizes it can sometimes wind up
		** with indices that are off the end of the block array.
		** So, we protect against that here, since this is the only callsite of mapping_search.
		** Note that we don't need to check sl, since it comes from a modulo operation that guarantees it's always in range.
		*/
		if (fl < control->fl_index_count)
		{
			block = search_suitable_block(control, &fl, &sl);
		}
	}

	if (block)
	{
		tlsf_assert(block_size(block) >= size && "block size must be larger than requested size");
		remove_free_block(control, block, fl, sl);
	}

	return block;
}

```

and here the changed Basic_HWPwm:
```
/*
 * File: Esp SDK Hardware PWM demo
 * Original Author: https://github.com/hrsavla
 *
 * This HardwarePWM library enables Sming framework user to use ESP SDK PWM API
 * Period of PWM is fixed to 1000us / Frequency = 1khz
 * Duty at 100% = 22222. Duty at 0% = 0
 * You can use function setPeriod() to change frequency/period.
 * Calculate the Duty as per the formulae give in ESP8266 SDK
 * Duty = (Period *1000)/45
 *
 * PWM can be generated on up to 8 pins (ie All pins except pin 16)
 * Created on August 17, 2015, 2:27 PM
 *
 * See also ESP8266 Technical Reference, Chapter 12:
 * http://espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf
 */
#include <SmingCore.h>
#include <HardwarePWM.h>

#define LED_PIN 2

namespace
{
// List of pins that you want to connect to pwm
#ifdef ARCH_ESP8266
uint8_t pins[]{
	LED_PIN, 4, 5, 0, 15, 13, 12, 14,
};
#endif
#ifdef ARCH_ESP32
uint8_t pins[]{
	3,4,5,18,19
};
#endif
//HardwarePWM HW_pwm(pins, ARRAY_SIZE(pins));
HardwarePWM* HW_pwm;

SimpleTimer procTimer;

int maxDuty ;

void doPWM()
{

	static bool countUp = true;
	static int duty,pin;

	const int increment = maxDuty / 50;

	if(countUp) {
		duty += increment;
		if(duty >= maxDuty) {
			duty = maxDuty;
			countUp = false;
		}
	} else {
		duty -= increment;
		if(duty <= 0) {
			duty = 0;
			countUp = true;
			HW_pwm->analogWrite(pins[pin], duty);
			pin+=1;
			if(pin == ARRAY_SIZE(pins))
				pin=0;
			debug_i("next pin: %i",pins[pin]);
		}
	}
	debug_i("   duty: %i", duty);
	HW_pwm->analogWrite(pins[pin], duty);
}

} // namespace

void init()
{
	Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
	Serial.systemDebugOutput(true); // Enable debug output to serial

	// WIFI not needed for demo. So disabling WIFI.
#ifndef DISABLE_WIFI
	WifiStation.enable(false);
	WifiAccessPoint.enable(false);
#endif
	debug_i("allocating HW_pwm: \n    number of pins: %i\n   size of array: %i\n   size of pin: %i",ARRAY_SIZE(pins), sizeof(pins), sizeof(pins[0]));
	HW_pwm = new HardwarePWM(pins, ARRAY_SIZE(pins));
	maxDuty= HW_pwm->getMaxDuty();
	debug_i("maxDuty: %i",HW_pwm->getMaxDuty());

	procTimer.initializeMs<100>(doPWM).start();
}
```
this compiles and works, as said, on Esp8266 and Esp32 but fails on Esp32c3. 
I don't think it's a Sming issue either, as the failure happens upon allocating the object, the constructor is never called. 

```
�114456 allocating HW_pwm: 
    number of pins: 5
   size of array: 5
   size of pin: 1

assert failed: block_locate_free tlsf.c:566 (block_size(block) >= size)
Core  0 register dump:
MEPC    : 0x40380698  RA      : 0x40383cf6  SP      : 0x3fc94730  GP      : 0x3fc8b400  
TP      : 0x3fc8be54  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x3fc94897  A0      : 0x3fc94788  A1      : 0x3fc8b48d  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc8d000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000080  S3      : 0x3fc94788  
S4      : 0x3fc94788  S5      : 0x3fc8ce44  S6      : 0x3fc8ce9c  S7      : 0xc0380000  
S8      : 0x00060000  S9      : 0x00000000  S10     : 0x3fc8d1e0  S11     : 0x3fc8ce30  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc94730: 0xffffffff 0xffffffff 0x3c026e4c 0x40389670 0xffffffff 0xffffffff 0xffffffff 0x3fc8b48c
3fc94750: 0x00363635 0xffffffff 0x00000000 0x00000000 0xff000000 0x3fc8b268 0x3c026e4c 0x3fc8b278
3fc94770: 0x3c02212b 0x3fc8b27c 0x3fc94750 0x3fc8b280 0x3c022364 0x3fc8b48c 0x65737361 0x66207472
3fc94790: 0x656c6961 0x62203a64 0x6b636f6c 0x636f6c5f 0x5f657461 0x65657266 0x736c7420 0x3a632e66
3fc947b0: 0x20363635 0x6f6c6228 0x735f6b63 0x28657a69 0x636f6c62 0x3e20296b 0x6973203d 0x0029657a
3fc947d0: 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff
3fc947f0: 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff
3fc94810: 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff 0xffffffff
3fc94830: 0xffffffff 0xffffffff 0xffffffff 0x0031ffff 0x00000069 0xffffffff 0x00000000 0x00000020
3fc94850: 0x00000000 0x3fc94889 0x3fc8ce68 0xa04412cd 0x0000000c 0x0000005c 0x3fc951f0 0x40388b64
3fc94870: 0x3fc94a34 0x00000000 0x00000000 0xc0380000 0x00000000 0x3fc8d000 0x00060000 0x00000000
3fc94890: 0x00000060 0x0000005c 0x3fc8ce30 0x40388500 0xffffffff 0xffffffff 0x3fc948c8 0x0000005c
3fc948b0: 0x3fc948bc 0x0000005c 0x00000804 0x40380ce6 0x00000000 0x00000000 0x00000058 0x00000000
3fc948d0: 0x00000000 0x00000000 0x00000000 0x0000000a 0x3fc8ac3c 0x00001388 0x0000000a 0x00000001
3fc948f0: 0x00000804 0x0000005c 0x3fc8cdf4 0x40380d00 0x00000001 0x00000000 0x3fc8cdf4 0x40384090
3fc94910: 0x6e697020 0x35203a73 0x2020200a 0x3fc8cdf4 0x00000001 0xffffffff 0x3fc8cdf4 0x403843b4
3fc94930: 0x657a6973 0x20666f20 0x3fc8cdf4 0x40380ff8 0x00000000 0x0000040e 0x0000000a 0x00000001
3fc94950: 0x0000040e 0x00000008 0x00000000 0x40381068 0x00000000 0x00000000 0x003e0000 0x00000000
3fc94970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00001388
3fc94990: 0x3fc8d000 0x3fc8d000 0x3fc949d8 0x4200d394 0x00000000 0x3fc951e4 0x3fc949d8 0x4200dd42
3fc949b0: 0x00000014 0x00000010 0x3fc8ac3c 0x00000000 0x00000005 0x3fc951e4 0x00000000 0x4200be1a
3fc949d0: 0x00000000 0x00000010 0x00000000 0x0000000a 0x00000000 0x00001388 0x00000000 0x00000000
3fc949f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94a10: 0x00001800 0x00000010 0x00000010 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94a30: 0x00000000 0x00000000 0x3fc951e4 0x4200442a 0xffffffff 0xffffffff 0x61207525 0x636f6c6c
3fc94a50: 0x6e697461 0x57482067 0x6d77705f 0x200a203a 0x6e202020 0x65626d75 0x666f2072 0x6e697020
3fc94a70: 0x25203a73 0x20200a69 0x7a697320 0x666f2065 0x72726120 0x203a7961 0x200a6925 0x69732020
3fc94a90: 0x6f20657a 0x69702066 0x25203a6e 0x000a0d69 0x00000000 0x00000000 0x3fc94ca4 0x4200420e
3fc94ab0: 0x00000000 0x00001f40 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x40384a7e
3fc94ad0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94af0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x0000015c 0x3fc94a10
3fc94b10: 0x00000000 0x3fc8be9c 0x3fc8be9c 0x3fc94b0c 0x3fc8be94 0x00000005 0x00000000 0x00000000

```

I will have to get out the jtag debugger.

mikee47, 2024-11-01:
Running the above modified Basic_HwPWM sample with your `fix/maxPWMChannels` branch works fine for me. It does throw the above if I attempt to run it against develop though.

pljakobs, 2024-11-02:
argh! I was fooled by this being a very different point in code (which makes sense since it's IDF) and also thought the patch had already been merged.
But what I don't understand: my fix only adds an assert - and given the configuration here, that should not be triggered, since we're only passing five channels. The assert is not being triggered.

But alas, you're right, the patch fixes it.

closed 2024-11-02 by pljakobs