SmingHub/Sming #2917: Support for Raspberry Pi Pico v2

slaff, 2024-11-26:
The new [Raspberry Pi Pico v2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/?variant=pico-2-w) seems to come with Wireless capabilities (Bluetooth and WiFi). This ticket is created to gather the requirements needed to port Sming for Pi Pico v2.

slaff, 2024-11-26:
@mikee47 any idea how much effort it will be to port to Pico V2 with wireless?

slaff, 2024-11-26:
>  I'll order up a few for testing.

If you want please, keep the bill(s) and I will reimburse you via OpenCollective.

mikee47, 2024-11-26:
@slaff Got an invoice, how best to get it to you?

mikee47, 2024-11-26:
I think I've had enough of esp32 for the time being. I'll take a look at updating the pico SDK for a pico-w.

slaff, 2024-11-26:
> Got an invoice, how best to get it to you?

Check your mike AT sillyhouse.net email

mikee47, 2024-11-27:
My intention is that building for the RP2350 family, including pico-2 or 2w, would be made using `SMING_SOC=rp2350`.

I note from the datasheet that the RP2350 has an A and B variant, the latter bringing out additional I/O. The SDK expects either `PICO_RP2350A` or `PICO_RP2350B` to be defined. The B variant is currently only defined for the 'hellbender' board.

I'll put the full set of pins in pindefs for now.

There's also the RP2354 with in-package flash, but it's a separate die so functionally equivalent to RP2350 with external 2MiB flash chip.

mikee47, 2024-11-27:
The other fun thing about the RP2350 is the configurable core architecture (ARM or RISCV). Let's start with ARM as the default, then we can add support for `PICO_RISCV=1` to select the other one.

mikee47, 2024-11-28:
Pre-built picotool (and pioasm) available as part of SDK tools here https://github.com/raspberrypi/pico-sdk-tools/releases
Could use those instead of building, but they're very small utilities so building would be my preference. Main issue is Windows of course.

mikee47, 2024-11-28:
The RP* microcontrollers are all flashed via USB so I find an additional USB/UART adapter for debugging, etc. is necessary.

RP2350 has a UART boot mode, described as *a minimal interface for bootstrapping a flashless RP2350 from another microcontroller*.

I thought this might offer a way to develop with a single serial port, but this mode uses QSPI SD2/3 pins so not useable on most dev. boards.

It might be that I'm using it all wrong and SWD offers a better interface for general development.

mikee47, 2024-11-28:
@slaff Do you have any rp* devices yourself?

slaff, 2024-11-28:
> Do you have any rp* devices yourself?

Not at the moment. Have to see if I can buy and get delivered quickly one in Germany.

closed 2024-12-04 by slaff