SmingHub/Sming #2926: Procedure to compile esp-idf example into Sming ?

profjmer, 2024-12-21:
I can compile and flash sdspi example in esp-idf, and it works.
If I create a new project in Sming with the same application, the compiler does not find some headers.
I did not find an easy way to deal with headers.

Do you have a procedure to import esp-idf examples into Sming projects?

BTW I tried Sming SDCard sample, but the spi lib used can't change spi pins and on some ESP32 boards the SD card interface needs different pins

Thank you.

mikee47, 2024-12-21:
There is no specific procedure to import esp-idf examples, as Sming is intended to keep the application separated from the architecture in use. The main complication with missing headers, etc. is because the IDF framework code is built as a static library rather than using the IDF build system to build the application itself. So if there is stuff you need from the IDF which is missing for your particular application then some modifications to the build logic in the `esp32` Component may be required.

The `SDCard` example should work with different pins, it just requires an extra line in the `sdInit()` function to call `SPI.setup()`. For example:

```
SPI.setup(SpiBus::DEFAULT, {.sck=MY_SCK_GPIO, .miso=MY_MISO_GPIO, .mosi=MY_MOSI_GPIO});
```

Note that this example uses the `SDCard` arduino library and requires use of the specific FAT access functions such as `f_open`, etc.

The `SdStorage` library provides an integration with the standard Sming filing system (IFS). The `Basic_IFS` example can be built  with `ENABLE_SDCARD=1`. You may need to change the `PIN_CARD_CS` definition in `application.cpp`, and also add the appropriate `SPI.setup()` call before line 228 (the call to `card->begin()`.

Both of these approaches uses the basic `SPI` library. If you require use of the ESP32 SDIO hardware then that will require use of the IDF directly.

profjmer, 2024-12-21:
Thank you for your explanation and Merry Christmas.

closed 2024-12-21 by profjmer