SmingHub/Sming #2936: possible division by zero in lwip/core/ipv4/igmp.c

pljakobs, 2025-01-23:
A user reported that his controllers were regularly restarting. 
Some troubleshooting revealed the following:

A WiFi access point in his network regularly sends the following query:
```
[Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/ipv4/igmp.c:444] igmp_input: General IGMP_MEMB_QUERY on "ALL SYSTEMS" address (224.0.0.1) [igmp_maxresp=1]
```

in processing this packet, the controller would crash 

```
***** Fatal exception 0 (ILLEGAL)
pc=0x4000e296 sp=0x3ffffe30 excvaddr=0x00000000
ps=0x00000030 sar=0x00000008 vpri=0x616c6564
r00: 0x40220148=1075970376 r01: 0x3ffffe30=1073741360 r02: 0x15cbcab2= 365677234 
r03: 0x00000000=         0 r04: 0x6000001c=1610612764 r05: 0x00000002=         2 
r06: 0x60000010=1610612752 r07: 0x00000000=         0 r08: 0x00303830=   3160112 
r09: 0x00000000=         0 r10: 0x3fff0960=1073678688 r11: 0x00000000=         0 
r12: 0x3fff4968=1073695080 r13: 0x00000001=         1 r14: 0x3fff4180=1073693056 
r15: 0x3fff4968=1073695080 
```

some digging in the code got me to the `igmp_start_timer` function
```
/**
 * Start a timer for an igmp group
 *
 * @param group the igmp_group for which to start a timer
 * @param max_time the time in multiples of IGMP_TMR_INTERVAL (decrease with
 *        every call to igmp_tmr())
 */
static void
igmp_start_timer(struct igmp_group *group, u8_t max_time)
{
  ip_addr_debug_print(IGMP_DEBUG, &group->group_address);
  LWIP_DEBUGF(IGMP_DEBUG, ("igmp_start_timer for %i",max_time));
  /* ensure the input value is > 0 */
  /* actually, it needs to be > 1 as 1 is subtracted before the modulo */
  if (max_time == 0 ) {
    max_time = 1;
  }
  /* ensure the random value is > 0 */
  group->timer = (LWIP_RAND() % (max_time - 1)) + 1;
}
```

the way this function is called in this code path is with `max_time=igmp->igmp_maxresp`

in setting the timer, `(max_time - 1)` is used as the divisor for the modulo operation, which results in a division by 0.

I have provided the user with a version where max time is required to be >1 and set to 2 if it is not, if that fixes the issue, I can submit a patch.

pljakobs, 2025-01-24:
actually, this is the upstream code from lwip

```
static void
igmp_start_timer(struct igmp_group *group, u8_t max_time)
{
#ifdef LWIP_RAND
  group->timer = (u16_t)(max_time > 2 ? (LWIP_RAND() % max_time) : 1);
#else /* LWIP_RAND */
  /* ATTENTION: use this only if absolutely necessary! */
  group->timer = max_time / 2;
#endif /* LWIP_RAND */

  if (group->timer == 0) {
    group->timer = 1;
  }
}
```
this avoids the div 0 condition

pljakobs, 2025-01-24:
happily, I guess since the esp-open-lwip subtree is a submodule with an external, rather stale repo, the PR should be against the patch file instead?

slaff, 2025-01-27:
@pljakobs please, check if the issue is fixed in latest `develop` and close the ticket if yes.

pljakobs, 2025-01-27:
I have requested feedback from the user having reported the issue. I have no way to reproduce this.

closed 2025-01-27 by pljakobs