SmingHub/Sming #2941: ESP32 wake up from external pin ?

profjmer, 2025-01-29:
Is there a way to set ESP32 to wake up from DeepSleep using for example GPIO32 pin (with pullup) from a low level?

mikee47, 2025-01-31:
Sming doesn't abstract this at present, but you should be able to use IDF SDK calls to do this.

profjmer, 2025-01-31:
How to make an IDF SDK call in Sming ?
Thanks

mikee47, 2025-01-31:
I've yet to make use of the esp32 sleep modes so the following is just from a casual inspection of the available docs. Hopefully it'll get you started.

Looking at the IDF `examples/system/deep_sleep` is perhaps a good starting point. From the [documentation](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32/api-reference/system/sleep_modes.html) the ESP32 has two external wakeup sources but no general GPIO. However, the technical reference manual says it does, but only for RTC GPIO pins.

The example contains code for using `ext0`, and the relevant code from `ext_wakeup.c` is this:

```
#include <esp_sleep.h>
#include <driver/rtc_io.h>

#if CONFIG_IDF_TARGET_ESP32
const int ext_wakeup_pin_0 = 25;
#else
const int ext_wakeup_pin_0 = 3;
#endif

void example_deep_sleep_register_ext0_wakeup(void)
{
    debug_i("Enabling EXT0 wakeup on pin GPIO%d", ext_wakeup_pin_0);
    ESP_ERROR_CHECK(esp_sleep_enable_ext0_wakeup(ext_wakeup_pin_0, 1));

    // Configure pullup/downs via RTCIO to tie wakeup pins to inactive level during deepsleep.
    // EXT0 resides in the same power domain (RTC_PERIPH) as the RTC IO pullup/downs.
    // No need to keep that power domain explicitly, unlike EXT1.
    ESP_ERROR_CHECK(rtc_gpio_pullup_dis(ext_wakeup_pin_0));
    ESP_ERROR_CHECK(rtc_gpio_pulldown_en(ext_wakeup_pin_0));
}
```

I added this to a new source file `ext_wakeup.cpp` in the `Basic_Blink` sample and it compiles.

profjmer, 2025-01-31:
Thanks, as always, your're very helpfull

closed 2025-01-31 by profjmer