SmingHub/Sming #2943: Timer on ESP32 hangs the chip if used with WiFi and a too fast rate for the Timer (ex 15mS), not happening on ESP8266.

profjmer, 2025-02-05:
On a ESP8266 I have an app, that is running an FSM by a Timer each 100uSec. (running fine with: Wifi, webserver, i2c, spi ...)

On an ESP32 a single bit toggle by a Timer can work at the same rate in Basic_Blink, but in Basic_Wifi sample, a rate faster than 15mS will hang the system.

Here is an extract of the test code:

	int i;
	IRAM_ATTR void systick()
	{
		if(i++>9){
			digitalWrite(22,!digitalRead(22));  // this is an LED on my ESP32 Board TTGO V1.3
			i=0;
		}
	}

in void init()
	pinMode(22,OUTPUT);
	pulse_timer.initializeMs<15>(systick).start();

ESP8266 has only one core and is performing better with Timer and WiFi
Esp32 has two cores, but Timer seems to interfere with WiFi at a rate too fast.
Is it something that can be improved in ESP32 ? (by not using the same core that WiFi uses ? or else)

mikee47, 2025-02-05:
Fast timers are very inefficient on the esp32 due to the interrupt and software servicing overhead. Perhaps look for a way to use the hardware directly, e.g. LEDC or other modules.

Sming builds with `CONFIG_FREERTOS_UNICORE=y` by default. See #2916 et. al.

mikee47, 2025-02-05:
NB. Are you using `HardwareTimer`? This has a lower software overhead as the callback is invoked in interrupt context. Software timers use the task queue which slows things down.

mikee47, 2025-02-05:
See also https://github.com/SmingHub/Sming/issues/2445#issuecomment-2485307349

closed 2025-02-05 by profjmer

mikee47, 2025-02-07:
@profjmer Be interested to hear about any solution you come up with for this.

profjmer, 2025-03-13:
I tested different ways with no great result. As Sming runs on unicore, I
guess controlling hardware with core 1 would do a good job, but right now I
can't figure out how to compile code for core 1 and start running it from a
Sming project. I'll let you know if I find a way to achieve that.

profjmer, 2025-03-15:
I'd like to test two unicore apps but on different cores.Is it possible to
specify core on which Sming will run ? Then cores would run separately but
simultaneously.

profjmer, 2025-03-15:
Concerning hardware timers, when interrupt handler is service to often,
freertos tasker kill the app

mikee47, 2025-03-16:
> I'd like to test two unicore apps but on different cores.Is it possible to
specify core on which Sming will run ? Then cores would run separately but
simultaneously.

Sming is started on the default application CPU via its implementation of `app_main()` in `Sming/Arch/Esp32/Components/esp32/src/startup.cpp`.

Run `make sdk-menuconfig` and setting `FREERTOS_UNICORE=n` to enable the second CPU. You then need to create a task running on the second CPU via `xTaskCreatePinnedToCore`:

```
void app2_main()
{
  // Code to run on second CPU
}

xTaskCreatePinnedToCore(app2_main, "Sming2", ESP_TASK_MAIN_STACK, nullptr, ESP_TASK_MAIN_PRIO, nullptr, 1);
```

Bear in mind that Sming is generally not thread-safe so app2 code would need to be written very carefully.