SmingHub/Sming #2952: Something wrong with I2C

teicors, 2025-04-18:
Hi,
I would connect a module I2C to ESP8266 SDA pin 2 and SDL pin 0 (D4 and D3 in my board). The sketch in Arduino works as expected, two sensors: BMP280 and AHT20.

Humidity: 64.07% rH
Temperature = 20.96 *C
Pressure = 99728.34 Pa
Approx altitude = 133.79 m

the test in Basic_Scanner_I2C shows

Scanning...
I2C device found at address 0x38  !
I2C device found at address 0x77  !
2 I2C devices found.

but when I run the Humidity_BME280, I'm having this error:

BME280 test
44124 [TWI] pins(2, 0)
44543 [TWI] begin(2, 0)
Could not find a valid BME280 sensor, check wiring, address, sensor ID!
SensorID was: 0x58
  ID of 0xFF probably means a bad address, a BMP 180 or BMP 085
  ID of 0x56-0x58 represents a BMP 280,
  ID of 0x60 represents a BME 280.
  ID of 0x61 represents a BME 680.

I cannot find my fault :(

mikee47, 2025-04-18:
You have a `BMP280` connected, but the same (and library) are for the `BME280`.

closed 2025-04-19 by teicors

mikee47, 2025-04-19:
I've opened PR #2953 to add the BMP280 support library to Sming.